PyCubed-Mini/avionics-solarpanels #8: Battery harness redesign

jbwillis, 2022-02-15:
The current battery harness is bulky and time consuming to build, mostly due to having to route wires between each set of batteries. A revision of the battery harness, possibly including some small PCBs, would allow for easier construction and testing of the batteries.

jbwillis, 2022-02-22:
Some more thoughts and questions on how to do this.

- [ ] Could we use only one type of battery? The more square ones are 150 mAh and seem to fit into the space better.
- [ ] Solder two batteries to a little circuit board that plugs into the -Y board?

jbwillis, 2022-02-23:
Another idea - solder two batteries to the bottom half of each of the X and Z boards, getting rid of the battery donut entirely. This would require adding another power signal to the FFC to take power from the batteries to the sep switches. Bad idea? maybe.

NeilKhera, 2022-08-31:
Solved with Revision B

closed 2022-08-31 by NeilKhera