PyCubed-Mini/avionics-solarpanels #9: Z Board access hole resize

jbwillis, 2022-02-22:
Need to make the access hole larger to enable easier access to USB port and to allow access to camera USB port.

- [ ] can we remove the fiducials on the z board and shift the solar panels all the way to the edge of the PCB?
- [ ] Extend hole down to even with solar cell below the hole.
- [ ] Make hole wider in each direction.

jbwillis, 2022-02-23:
This is also going to impact the magnetorquer coil routing on the Z board. However, I noticed that the Z coils stay within the mounting holes, whereas the X wrap around the outside of the mounting holes. So, the Z coils could wrap outside the holes, which would reduce the impact of the access hole.

NeilKhera, 2022-02-28:
I don't love the idea of have a big hole on two sides of the sat. Seems like a bad idea for contaminants and such entering the inner boards? Could we instead try to have some sort of external utility port which we can hook up to and breakout whatever we need?

jbwillis, 2022-02-28:
That's an interesting idea. It's pretty standard for cubesats to have one of these holes. I don't think there is a huge contaminant problem for us - I could see that being an issue for something like MoonRanger though. We also already have a massive hole that the camera protrudes through, so I'm not sure how much we'd gain by avoiding that on the Z board. Adding an external interface also adds extra complications for assembly and testing. Let's discuss this with @zacmanchester to get his final word.

closed 2022-03-16 by NeilKhera